GradientSpaces/WildGS-SLAM #3: Custom dataset

aloma85, 2025-04-09:
Awesome work! got it to run with minor errors. Can you give a quick breakdown on how to run with custom dataset? I have just the video frames with no ground truth depth, trajectories, and intrinsics.

Btw anyone getting

> cv2.error: OpenCV(4.8.1) /io/opencv/modules/calib3d/src/undistort.dispatch.cpp:299: error: (-215:Assertion failed) dst.data != src.data in function 'undistort'

Just loop this line until the image is recognized

`color_data_fullsize = cv2.undistort(color_data_fullsize, K, self.distortion)`

Jianhao-zheng, 2025-04-10:
Hi! Thanks for your interest in our work. For custom dataset with only RGB frames, I would recommend you to refer to how we do the config for wild-slam-iphone dataset. Typically, you need to set `dataset` to `wild_slam_iphone` and modify the camera parameters (see [here ](https://github.com/GradientSpaces/WildGS-SLAM/blob/eab7a3d6ebe2fe3c50d0b221eb4db2ab0b0e399b/configs/Dynamic/Wild_SLAM_Mocap/wild_slam_mocap.yaml#L9)for reference) in the config file.

The error you countered here might because you didn't load the image from the correct path. Can you check whether `color_data_fullsize` is None?

robinlindh, 2025-04-10:
Btw @aloma85 if you try the iPhone dataset, notice that the dataset path is currently set to an absolute path so you need to change that. I just now opened a PR for this small fix #4

Jianhao-zheng, 2025-04-10:
This issue of the dataset path was actually fixed in eab7a3d6ebe2fe3c50d0b221eb4db2ab0b0e399b

Hamme122, 2025-04-11:
I recommend you build a `.yaml` file under `./configs/Dynamic/Wild_SLAM_iPhone`, just like this:

```
inherit_from: ./configs/Dynamic/Wild_SLAM_iPhone/wild_slam_iphone.yaml
scene: iphone_myscene

data:
  input_folder: /nas/Data/Wildslam/Wild_SLAM_iPhone/myscene

cam:
  fx: 370.5449868115882
  fy: 366.1343254037455
  cx: 180
  cy: 240
  H: 360 
  W: 480 
  H_out: 360
  W_out: 480
```

Notice that you need to fill the './myscene/rgb' file folder with your images and get the info of the cam intrinsic.

closed 2025-04-18 by aloma85