GradientSpaces/WildGS-SLAM #5: Some questions about the ground truth images

Hamme122, 2025-04-10:
[image]

When I reproduced some results from the iPhone datasets, I noticed that some images will have unnatural blur and wave surrounding the light source. But the input images is correct. Could you explain the possible reason? Thank you!

Jianhao-zheng, 2025-04-10:
I think your screenshot here is from the gif. That's due to the loss of quality when compressing the gif. If you check the .png image files, they are correct.

Hamme122, 2025-04-11:
> I think your screenshot here is from the gif. That's due to the loss of quality when compressing the gif. If you check the .png image files, they are correct.

Thanks for your reply. It is indeed a problem in GIF, and it will be better if we transform it to video.

closed 2025-04-11 by Hamme122