GradientSpaces/WildGS-SLAM #6: Customized Dataset and Long Runtime Problems

zhuoranli23, 2025-04-18:
Hello, WildGS-SLAM team! Thank you for your awesome work!!

I am very new to this field (SLAM, 3DGS, and 3D reconstruction) and would like to ask you some questions. I appreciate any help you can provide.

**1. How to run/use WildGS-SLAM on my customized dataset？**
I am doing relevant research, and our team has a personalized dataset that only has RGB videos/image frames. Since WildGS-SLAM only requires RGB image frames as input, how should I use it on our dataset? **Specifically**, **(i)** which parts of this repo code base need to be modified/changed accordingly, **(ii)** are RGB images enough, **(iii)** are depth images required or not (From my understanding, you used depth images just for quantitative experiments only, not for the method pipeline. Is my understanding correct?), **(iv)** are camera parameters required/camera calibration required?

**2. What is the estimated training/runtime for the code?**
I tried your Quick Demo by `python run.py  ./configs/Dynamic/Wild_SLAM_Mocap/crowd_demo.yaml`, and my PC has one RTX4080 GPU and one i9 Intel CPU. But the terminal was stuck as follows for several hours. After reading the paper, I came to know that the model is trained on-the-fly (i.e., online training) for the image frames, so the long runtime means the model was still training online. Is my understanding correct? **I would like to know what the estimated training/runtime is for this quick demo, and for a regular dataset that has around 2000 frames?** 
> Start WildGS-SLAM at 2025-04-18 14:02:03,
>    scene: wild_slam_mocap-crowd_demo,
>    output: ./output/Wild_SLAM_Mocap_demo/crowd_demo
> ------------------------------
> /home/zhuoran/anaconda3/envs/wildgs-slam/lib/python3.10/site-packages/torch/_utils.py:831: UserWarning: TypedStorage is deprecated. It will be removed in the future and UntypedStorage will be the only storage class. This should only matter to you if you are using storages directly.  To access UntypedStorage directly, use tensor.untyped_storage() instead of tensor.storage()
>   return self.fget.__get__(instance, owner)()
> [INFO] Load droid pretrained checkpoint from ./pretrained/droid.pth!
> /home/zhuoran/anaconda3/envs/wildgs-slam/lib/python3.10/site-packages/torch/hub.py:294: 
.......
> Downloading: "https://huggingface.co/yuanwenyue/FiT3D/resolve/main/dinov2_reg_small_finetuned.pth" to /home/zhuoran/.cache/torch/hub/checkpoints/dinov2_reg_small_finetuned.pth
> 100%|██████████████████████████████████████████████████████| 84.2M/84.2M [00:03<00:00, 22.2MB/s]
> [TRACKER] Tracking Triggered! 

Thank you very much for your kind help, and I really appreciate it!!

aloma85, 2025-04-18:
[Here's how to use a custom dataset](https://github.com/GradientSpaces/WildGS-SLAM/issues/3)

Jianhao-zheng, 2025-04-18:
Hi thank you for your interest in our work.

1. (i) You can refer to any `yaml` file under the `./configs/Dynamic/Wild_SLAM_iPhone` folder for your customized dataset. You need to extract your images from the videos and put it under `{your_data_folder}/rgb/`. In the ymal file, you need at least to specify the `input_folder`, `scene`, and `cam` (the camera intrinsics). (ii) Yes, our work is a "monocular" SLAM system and it requires RGB images only. (iii) We provide depth images in our dataset for other methods requiring depth frames as input. We don't need any depth info to run the SLAM system nor the evaulation. (iv) intrinsic calibration is needed. If you don't have that, you can try some other methods to estimate it (e.g. Colmap, Monst3r).
2. We did the experiments in a computer with 4090 and the running time is around 0.5 FPS (full pipeline) and 2 FPS (fast mode, which is activated in the quick demo). These data and more values can be found in the supplementary pages of our paper. But in your case, you said it stuck for several hours. There might be something wrong. Note that the network trained online is just a two-layer MLP, you should be able to see the map and pose trajectory keep changing when you run the demo.

Jianhao-zheng, 2025-06-03:
I have added the instructions to run WildGS-SLAM with your own dataset. Close this issue since no update for a whle.

closed 2025-06-03 by Jianhao-zheng