GradientSpaces/WildGS-SLAM #8: Why is it saving a frame every 10 stitches? How can I save each rendered frame?

lichengyang-robot, 2025-04-19:
Great work! Why is it saving a frame every 10 stitches? How can I save each rendered frame?

Jianhao-zheng, 2025-04-21:
Hi, thank you for your interest in our work. We first do tracking and mapping on selected keyframes, then fill in the pose for non-key frames. This is similar to Splat-SLAM. If you want to do mapping with all the frames, you need to change the code. Or you can force to add every frame as a keyframe by setting the following parameters:
```
tracking:
  force_keyframe_every_n_frames: 1 # Default value is 9
```
**Note that long sequences are not recommended, as this will require lots of memory. Also, don't forget to set the buffer size to be greater than your sequence length. #7** 

If you only want to see the rendering of non-keyframes instead of optimizing the map with them, please react with a thumbs-up, and I will add code related to this later on.

Jianhao-zheng, 2025-05-07:
Closed as no more updates.

closed 2025-05-07 by Jianhao-zheng