GradientSpaces/WildGS-SLAM #9: Image Resize problem

Hamme122, 2025-04-22:
Thanks for your outstanding work! I have some questions about the image resize in the .yaml file, such as:
```
cam:
  H: 1920 
  W: 1080 
  H_out: 1280
  W_out: 640
```

The original H_out and W_out are both below 512. If I set it into a larger number, it will has an error:

```
[TRACKER] :   5%|▍         | 28/579 [00:04<01:23,  6.56it/s]
[TRACKER] :   5%|▍         | 28/579 [00:04<01:28,  6.24it/s]
[TR...
Traceback (most recent call last):
  File "/cpfs04/user/miniconda3/envs/wildgs/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/cpfs04/user/miniconda3/envs/wildgs/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/cpfs04/user/Projects/WildGS-SLAM/src/slam.py", line 108, in tracking
    self.tracker.run(self.stream)
  File "/cpfs04/user/Projects/WildGS-SLAM/src/tracker.py", line 55, in run
    self.frontend(force_to_add_keyframe)
  File "/cpfs04/user/Projects/WildGS-SLAM/src/frontend.py", line 165, in __call__
    self.__initialize()
  File "/cpfs04/user/Projects/WildGS-SLAM/src/frontend.py", line 104, in __initialize
    self.graph.add_neighborhood_factors(0, self.t1, r=3)
  File "/cpfs04/user/Projects/WildGS-SLAM/src/factor_graph.py", line 322, in add_neighborhood_factors
    self.add_factors(ii[keep], jj[keep])
  File "/cpfs04/user/miniconda3/envs/wildgs/lib/python3.10/site-packages/torch/amp/autocast_mode.py", line 16, in decorate_autocast
    return func(*args, **kwargs)
  File "/cpfs04/user/miniconda3/envs/wildgs/lib/python3.10/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context
    return func(*args, **kwargs)
  File "/cpfs04/user/Projects/WildGS-SLAM/src/factor_graph.py", line 125, in add_factors
    corr = CorrBlock(fmap1, fmap2)
  File "/cpfs04/user/Projects/WildGS-SLAM/src/modules/droid_net/corr.py", line 55, in __init__
    corr = F.avg_pool2d(corr, kernel_size=2, stride=2)
RuntimeError: integer out of range
```

I have found it is a problem in the feature pyramid and one torch's multiple of shape has exceeded int32. But I haven't known how to resolve it. Could you give some suggestions in this problem? Thanks very much!

Jianhao-zheng, 2025-04-22:
Hi Hamme122, this is an interesting error. I have not seen this before. Looks like it's related to the code of the droid network. I think if you make the size of the output images smaller, this can be resolved. Note that tracking typically doesn't need a high resolution input. If you want to improve mapping with higher resolution images, we provide an option to use full full-resolution image for mapping [here](https://github.com/GradientSpaces/WildGS-SLAM/blob/eab7a3d6ebe2fe3c50d0b221eb4db2ab0b0e399b/configs/wildgs_slam.yaml#L11C3-L11C18).

Hamme122, 2025-04-22:
> Hi Hamme122, this is an interesting error. I have not seen this before. Looks like it's related to the code of the droid network. I think if you make the size of the output images smaller, this can [...]

Thanks for your reply! I will try the full-resolution mapping. Thanks!

closed 2025-04-22 by Hamme122