GradientSpaces/WildGS-SLAM #10: A question about uncertainty in the tracking process

EH-HU, 2025-04-24:
In the pipeline described in the paper, the predicted uncertainty is passed to the DBA module. Can you help me identify where this is handled in the code?

Jianhao-zheng, 2025-04-25:
Hi, it's main used here: https://github.com/GradientSpaces/WildGS-SLAM/blob/eab7a3d6ebe2fe3c50d0b221eb4db2ab0b0e399b/src/depth_video.py#L354

EH-HU, 2025-05-02:
After commenting out the specific line of code, I observed a slight improvement in RMSE performance. Could you explain the reason behind this result?

Jianhao-zheng, 2025-05-05:
If you want to see performance without uncertainty mask, please set `cfg['tracking']["uncertainty_params"]['activate']` to False. Also, can you provide the sequences you run?

EH-HU, 2025-05-06:
When I set cfg['tracking']["uncertainty_params"]['activate'] to False, I encountered the following error:
Exception: This function should not be called if uncertainty aware is not activated.
To bypass this, I commented out the following line of code:
weight *= self.uncertainties_inv[ii][None, :, :, :, None]
After doing so, I was able to run the test.
The sequence I used for testing was bonn_balloon2.

EH-HU, 2025-05-06:
Additionally, is there a specific reason why the code
does not overwrite with an assignment (=) but instead uses a multiplication update (*=)?

weight *= self.uncertainties_inv[ii][None, :, :, :, None]

Jianhao-zheng, 2025-05-07:
I have updated the code to fix the bug and you should be able to run it without error now. Bonn_balloon2 is a simple sequence for dynamic SLAM task. Even Splat-SLAM, which is designed for static SLAM, can achieves similar performance to WIldGS-SLAM on this sequence. Please try with some other sequences.

`weight` itself is one of the output from ConvGRU that will be used in Global Bundle Adjustment. We multiply it by the uncertainty_inv to further reduce the weight for dynamic pixels. It's not ideal to replace the weight with uncertainty_inv.

EH-HU, 2025-05-08:
hank you for your response.
I have confirmed that I obtained improved results on the bonn_crowd sequence.
In the full trajectory evaluation, I achieved an RMSE of 0.01689974868934451, which seems slightly different from the result reported in the paper.
Could this difference be due to variations in computer performance?

Jianhao-zheng, 2025-05-08:
yes

closed 2025-05-08 by Jianhao-zheng