GradientSpaces/WildGS-SLAM #11: Bus error (core dumped) when running  python run.py configs/Dynamic/Bonn/bonn_balloon2.yaml

qingshufan, 2025-05-04:
Bus error (core dumped) when running  python run.py configs/Dynamic/Bonn/bonn_balloon2.yaml 
 
Environment Information
 
- Operating System: Ubuntu 22.04 on a cloud server
 
- Python Environment: Python 3.10 within the  wildgs-slam  conda environment
 
Steps to Reproduce
 
1. Access the directory  /home/test/WildGS-SLAM/ .
 
2. Activate the  wildgs-slam  environment using the command  conda activate wildgs-slam .
 
3. Execute the command  python run.py configs/Dynamic/Bonn/bonn_balloon2.yaml .
 
Error Messages
 
Initially, a warning from PyTorch is displayed:
 
plaintext
  
/root/miniconda3/envs/wildgs-slam/lib/python3.10/site-packages/torch/_utils.py:831: UserWarning: TypedStorage is deprecated...
 
 
Subsequently, the  Bus error (core dumped)  error occurs.
Following that, the message  [INFO] Load droid pretrained checkpoint from ./pretrained/droid.pth!  appears, though its relation to the error is unclear.
 
Thank you for your attention to this issue. I would greatly appreciate it if you could provide any insights or suggestions on how to resolve this problem. If there is any additional information I can provide, please let me know.

Jianhao-zheng, 2025-05-05:
Hi, thank you for your interest in our work. I didn't encounter the "core dumped" error previously. It's also new for me. I suspect it might be related to the installation of lietorch. Could you try to run `python -c "import lietorch"` and see if there's any error raised? Otherwise, I think we need to find the error step by step. Can you provide the entire printed output to me?

qingshufan, 2025-05-06:
你好！感谢你的帮助。我按照你的建议运行了 `python -c "import lietorch"`，没有发现报错。不过运行 `python run.py configs/Dynamic/Bonn/bonn_balloon2.yaml` 时，还是出现了 `Bus error (core dumped)` 的错误。完整的输出如下：

```
python run.py configs/Dynamic/Bonn/bonn_balloon2.yaml
------------------------------
Start WildGS - SLAM at 2025-05-04 14:56:51,
   scene: bonn_dynamic-bonn_balloon2,
   output: ./output/Bonn/bonn_balloon2
------------------------------
/root/miniconda3/envs/wildgs-slam/lib/python3.10/site-packages/torch/_utils.py:831: UserWarning: TypedStorage is deprecated. It will be removed in the future and UntypedStorage will be the only storage class. This should only matter to you if you are using storages directly.  To access UntypedStorage directly, use tensor.untyped_storage() instead of tensor.storage()
  return self.fget.__get__(instance, owner)()
Bus error (core dumped)
(wildgs-slam) root@g45efuj9t92c-0:/home/test/WildGS-SLAM# [INFO] Load droid pretrained checkpoint from ./pretrained/droid.pth!
```

我自己经过长时间测试，发现把 `configs/wildgs_slam.yaml` 里 `mono_prior.depth` 的 `'metric3d_vit_large'` 改成 `'metric3d_vit_small'` 后，程序能够成功运行。

Hello! Thank you for your help. I ran `python -c "import lietorch"` as you suggested, and there were no error messages. However, when running `python run.py configs/Dynamic/Bonn/bonn_balloon2.yaml`, I still encountered the `Bus error (core dumped)` error. The full output is as follows:

```
python run.py configs/Dynamic/Bonn/bonn_balloon2.yaml
------------------------------
Start WildGS - SLAM at 2025-05-04 14:56:51,
   scene: bonn_dynamic-bonn_balloon2,
   output: ./output/Bonn/bonn_balloon2
------------------------------
/root/miniconda3/envs/wildgs-slam/lib/python3.10/site-packages/torch/_utils.py:831: UserWarning: TypedStorage is deprecated. It will be removed in the future and UntypedStorage will be the only storage class. This should only matter to you if you are using storages directly.  To access UntypedStorage directly, use tensor.untyped_storage() instead of tensor.storage()
  return self.fget.__get__(instance, owner)()
Bus error (core dumped)
(wildgs-slam) root@g45efuj9t92c-0:/home/test/WildGS-SLAM# [INFO] Load droid pretrained checkpoint from ./pretrained/droid.pth!
```

After extensive testing on my own, I found that changing `'metric3d_vit_large'` to `'metric3d_vit_small'` in `mono_prior.depth` of `configs/wildgs_slam.yaml` allows the program to run successfully.

closed 2025-05-06 by qingshufan

Jianhao-zheng, 2025-05-07:
Thank you for the update. I am glad you solved it. Just a note, the error indicates that the memory of your workstation is not enough to load the large version of metric3d model. Although changing it to small will make the pipeline run, the results might be worse than using large version.

qingshufan, 2025-05-08:
Thanks for letting me know. I totally get what you're saying. The model's loading prompt is like a riddle! It really should be more straightforward. But I'll manage with the small version for now. Thanks again!