GradientSpaces/WildGS-SLAM #15: tensorlist problem

Alan-zhong, 2025-05-19:
Hi! amazing work!
when i run it on my own custom datasets,it had somepeoblem
[image]

i create a new test.yaml 

[image]

Jianhao-zheng, 2025-05-19:
Hi, thank you for your interest in our work. From your output, I see no frames are processed. I guess it's maybe due to the incorrect path to your dataset? Another potential issue could be that your images are using formats other than ".png". Note that the no-pose dataloader will only search fro ".png" images.
https://github.com/GradientSpaces/WildGS-SLAM/blob/420a984c1f53607dd0afa3286c08f47a2ac3139a/src/utils/datasets.py#L420

You can simply change the above line to accept other formats.

Jianhao-zheng, 2025-05-29:
Hi, do you have any updates or further questions? If not, I will close this.

Jianhao-zheng, 2025-06-02:
close due to no update

closed 2025-06-02 by Jianhao-zheng