GradientSpaces/WildGS-SLAM #20: Depth L1 seems to have a huge error in unseen data.

truncs, 2025-05-29:
I am trying to evaluate the performance of WildGS-SLAM on unseen images and I am seeing that the rendered depth has huge errors as shown in the output gif. Any suggestions?

Jianhao-zheng, 2025-05-30:
Hi, thank you for your interest in our work. I tried some input sequences similar to yours and wildgs-slam should be able to render some reasonable results. I feel like it's maybe because the depth is not optimized well. I have a hard-coded parameter that will not optimize the L1 depth for pixel whose metric depth is more than 50

Can you try to set this number 50 higher or simply set ``depth_threshold=100``?

https://github.com/GradientSpaces/WildGS-SLAM/blob/420a984c1f53607dd0afa3286c08f47a2ac3139a/src/utils/dyn_uncertainty/mapping_utils.py#L257

If the results are still not good, can you share the input video to me (if possible)?

Jianhao-zheng, 2025-06-02:
Thanks, I will check it and get back to you.

truncs, 2025-06-05:
Interestingly my results are still bad. I tried with the custom config that you pushed and with the intrinsic setting. [link](https://drive.google.com/file/d/1EjHAOKpSORI7otK80yZ8qI_isT4vT9hi/view?usp=sharing)

The diff between the repos is not relevant and I am posting it over here and do note that even though I have the Sanpo configs I did run the above link using the Custom config from the latest commit.

```diff
diff --git a/configs/Custom/custom_template.yaml b/configs/Custom/custom_template.yaml
index 7ae0348..621e43e 100755
--- a/configs/Custom/custom_template.yaml
+++ b/configs/Custom/custom_template.yaml
@@ -3,7 +3,7 @@ scene: custom_scene # Replace with your scene name

 dataset: 'wild_slam_iphone'
 data:
-  input_folder: ./datasets/{Path_to_your_data}
+  input_folder: /home/nonroot/datasets/sanpo/-PqSDmiEe2pXjmYHgxh4YEBsj0T5LU10/camera_head/left/video_frames/
   output: ./output/Custom

 cam:
diff --git a/configs/Dynamic/Sanpo/config.yaml b/configs/Dynamic/Sanpo/config.yaml
new file mode 100644
index 0000000..881077a
--- /dev/null
+++ b/configs/Dynamic/Sanpo/config.yaml
@@ -0,0 +1,27 @@
+inherit_from: ./configs/wildgs_slam.yaml
+scene: '-PqSDmiEe2pXjmYHgxh4YEBsj0T5LU10'
+dataset: 'sanpo'
+gui: True
+
+data:
+  root_folder: ./datasets/Sanpo
+  output: ./output/Sanpo
+  input_folder: /home/nonroot/datasets/sanpo/-PqSDmiEe2pXjmYHgxh4YEBsj0T5LU10/camera_head/left/video_frames/
+
+cam:  #intrinsic is slightly different per seq
+  H: 1242
+  W: 2208
+  fx: 1974.4219
+  fy: 1974.4219
+  cx: 1134.8486
+  cy: 655.6515
+  H_out: 360
+  W_out: 480
+
+mapping:
+  Training:
+    alpha: 0.8 # Increase this value to make rendering loss weighs more on rgb rather than depth
+  uncertainty_params:
+    # For outdoor dataset where the metric depth estimation is unstable,
+    # I recommend to set this value to be 0.1 or even 0.
+    uncer_depth_mult: 0.2
diff --git a/docker/Dockerfile b/docker/Dockerfile
new file mode 100644
index 0000000..538ca7f
--- /dev/null
+++ b/docker/Dockerfile
@@ -0,0 +1,64 @@
+FROM nvidia/cuda:11.8.0-cudnn8-devel-ubuntu22.04
+ARG UID
+ARG GID
+
+RUN apt update && \
+    apt install -y sudo && \
+    addgroup --gid $GID nonroot && \
+    adduser --uid $UID --gid $GID --disabled-password --gecos "" nonroot && \
+    echo 'nonroot ALL=(ALL) NOPASSWD: ALL' >> /etc/sudoers
+
+RUN apt-get update -y && \
+    apt-get install -y wget software-properties-common && \
+    add-apt-repository -y ppa:ubuntu-toolchain-r/test && \
+    wget -O - https://apt.llvm.org/llvm-snapshot.gpg.key| apt-key add
+RUN apt-add-repository -y "deb http://archive.ubuntu.com/ubuntu focal main universe"
+
+ARG DEBIAN_FRONTEND=noninteractive
+
+RUN apt-get install -y tzdata && \
+    ln -fs /usr/share/zoneinfo/America/Los_Angeles /etc/localtime && \
+    dpkg-reconfigure --frontend noninteractive tzdata
+
+RUN apt-get install -y build-essential libgl1-mesa-glx \
+    python3 python python3-dev python3-pip libpng-dev libtiff5-dev \
+    libjpeg-dev tzdata sed curl wget unzip autoconf libtool rsync libxml2-dev \
+    git git-lfs
+
+RUN mkdir -p /opt/conda
+RUN wget https://repo.anaconda.com/miniconda/Miniconda3-py310_25.1.1-2-Linux-x86_64.sh -O /opt/conda/miniconda.sh && bash /opt/conda/miniconda.sh -b -p /opt/miniconda
+
+# Set the non-root user as the default user
+USER nonroot
+
+WORKDIR "/home/nonroot/"
+RUN git clone --recursive https://github.com/GradientSpaces/WildGS-SLAM.git
+
+WORKDIR "/home/nonroot/WildGS-SLAM/"
+RUN /opt/miniconda/bin/conda create --name wildgs-slam python=3.10
+RUN echo "conda activate wildgs-slam" > ~/.bashrc
+RUN echo "PATH=/opt/miniconda/envs/wildgs-slam/:/opt/miniconda/bin:$PATH"
+ENV PATH /opt/miniconda/envs/wildgs-slam/bin:$PATH
+SHELL ["/bin/bash", "--login", "-c"]
+
+RUN pip install numpy==1.26.3 # do not use numpy >= v2.0.0
+RUN pip install torch==2.1.0 torchvision==0.16.0 torchaudio==2.1.0 --index-url https://download.pytorch.org/whl/cu118
+RUN pip install torch-scatter -f https://pytorch-geometric.com/whl/torch-2.1.0+cu118.html
+RUN pip3 install -U xformers==0.0.22.post7+cu118 --index-url https://download.pytorch.org/whl/cu118
+
+RUN pip install -e thirdparty/lietorch/
+# RUN pip install -e thirdparty/diff-gaussian-rasterization-w-pose/
+# RUN pip install -e thirdparty/simple-knn/
+
+# RUN python -c "import torch; import lietorch; import simple_knn; import diff_gaussian_rasterization; print(torch.cuda.is_available())"
+
+# RUN python -m pip install -e .
+# RUN python -m pip install -r requirements.txt
+# RUN pip install gdown
+
+# RUN pip install mmcv-full -f https://download.openmmlab.com/mmcv/dist/cu118/torch2.1.0/index.html
+# RUN gdown 1PpqVt1H4maBa_GbPJp4NwxRsd9jk-elh
+# RUN mkdir -p pretrained
+# RUN mv droid.pth pretrained/
+
+
diff --git a/src/utils/datasets.py b/src/utils/datasets.py
index 50fa3f7..7a42a08 100644
--- a/src/utils/datasets.py
+++ b/src/utils/datasets.py
@@ -417,7 +417,7 @@ class RGB_NoPose(BaseDataset):
                  ):
         super(RGB_NoPose, self).__init__(cfg, device)
         self.color_paths = sorted(
-            glob.glob(f'{self.input_folder}/rgb/frame*.png'))
+            glob.glob(f'{self.input_folder}/*.png'))
         self.depth_paths = None
         self.poses = None

@@ -429,11 +429,13 @@ class RGB_NoPose(BaseDataset):
         self.color_paths = self.color_paths[:max_frames][::stride]
         self.n_img = len(self.color_paths)

+
 dataset_dict = {
     "replica": Replica,
     "scannet": ScanNet,
     "tumrgbd": TUM_RGBD,
     "bonn_dynamic": TUM_RGBD,
     "wild_slam_mocap": TUM_RGBD,
-    "wild_slam_iphone": RGB_NoPose
+    "wild_slam_iphone": RGB_NoPose,
+    "sanpo": RGB_NoPose,
 }
(END)
```

Jianhao-zheng, 2025-06-07:
I didn't see any difference that might influence the result. Can you try our demo ``python run.py  ./configs/Dynamic/Wild_SLAM_Mocap/crowd_demo.yaml``? Does the result look good? Also, I see you have activated GUI, what did you during online mapping? You can activate the fast_mode to get fast feedback.

Jianhao-zheng, 2025-06-27:
Close due to no response for a while.

closed 2025-06-27 by Jianhao-zheng