GradientSpaces/WildGS-SLAM #21: About "RuntimeError: CUDA error: invalid resource handle"

hyacinth531, 2025-06-02:
Hello, everyone! I have a question to ask. When I run wildsg-slam and it reaches“p.start()” , it prompts:
**"RuntimeError: CUDA error: invalid resource handle
Compile with TORCH_USE_CUDA_DSA to enable device-side assertions."**
I found that when I change torch.multiprocessing.set_start_method('spawn') to “fork”, this error will not occur. However, for wildsg-slam, the multithreading mode must be set to “spawn”.
I guess this is a problem that occurs during multithreaded execution. I wonder if anyone has encountered a similar problem and knows how to solve it. Thank you!
In addition, the CUDA installation is all correct.

Jianhao-zheng, 2025-06-03:
It's very weird. I cannot reproduce your error and I am also using 4090. Can you try [Splat-SLAM](https://github.com/google-research/Splat-SLAM)? Our multi-processing is based on their code.

hyacinth531, 2025-06-04:
I'm running it in WSL. I wonder if it has any impact.

Jianhao-zheng, 2025-06-04:
It could be the reason. If you have time, you can run Splat-Slam with a small test. When you encounter the same error, it's definitely an OS problem.

hyacinth531, 2025-06-07:
Thank you for your help.
I also ran Splat-SLAM and got the same error. I highly suspect that it is a problem with WSL. I had no issues when running it on the AutoDL online cloud. However, it seems that no results can be displayed. Since AutoDL doesn't seem to support display functions like X11, I'm wondering if there are any good solutions. Thank you!

hyacinth531, 2025-06-07:
I've found that someone has successfully run Wildgs-slam on WSL, although the speed is relatively slow. However, when I run both Wildgs-slam and Splat-SLAM on WSL with Ubuntu 20.04 and WSL with Ubuntu 22.04, I get the following error messages:

(wildgs-slam) hyacinth@wuming:/mnt/e/My_Program/WildGS-SLAM$ python run.py  ./configs/Dynamic/Wild_SLAM_Mocap/crowd_demo.yaml
------------------------------
Start WildGS-SLAM at 2025-06-07 12:32:32,
   scene: wild_slam_mocap-crowd_demo,
   output: ./output/Wild_SLAM_Mocap_demo/crowd_demo
------------------------------
/home/hyacinth/anaconda3/envs/wildgs-slam/lib/python3.10/site-packages/torch/_utils.py:831: UserWarning: TypedStorage is deprecated. It will be removed in the future and UntypedStorage will be the only storage class. This should only matter to you if you are using storages directly.  To access UntypedStorage directly, use tensor.untyped_storage() instead of tensor.storage()
  return self.fget.__get__(instance, owner)()
[INFO] Load droid pretrained checkpoint from ./pretrained/droid.pth!
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/hyacinth/anaconda3/envs/wildgs-slam/lib/python3.10/multiprocessing/spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "/home/hyacinth/anaconda3/envs/wildgs-slam/lib/python3.10/multiprocessing/spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
  File "/home/hyacinth/anaconda3/envs/wildgs-slam/lib/python3.10/site-packages/torch/multiprocessing/reductions.py", line 147, in rebuild_cuda_tensor
    storage = storage_cls._new_shared_cuda(
  File "/home/hyacinth/anaconda3/envs/wildgs-slam/lib/python3.10/site-packages/torch/storage.py", line 1085, in _new_shared_cuda
    return torch.UntypedStorage._new_shared_cuda(*args, **kwargs)
RuntimeError: CUDA error: invalid resource handle
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

The following is the error message that appears when running Splat-SLAM：

(splat-slam) hyacinth531@wuming:/mnt/e/My_Program/Splat-SLAM$ python run.py configs/Replica/office0.yaml
------------------------------
Start Splat-SLAM at 2025-06-07 12:38:07,
   scene: replica-office0,
   only_tracking: False,
   output: output/Replica/office0
------------------------------
[INFO] Load droid pretrained checkpoint from pretrained/droid.pth!
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/hyacinth531/miniconda3/envs/splat-slam/lib/python3.10/multiprocessing/spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "/home/hyacinth531/miniconda3/envs/splat-slam/lib/python3.10/multiprocessing/spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
  File "/home/hyacinth531/miniconda3/envs/splat-slam/lib/python3.10/site-packages/torch/multiprocessing/reductions.py", line 147, in rebuild_cuda_tensor
    storage = storage_cls._new_shared_cuda(
  File "/home/hyacinth531/miniconda3/envs/splat-slam/lib/python3.10/site-packages/torch/storage.py", line 1085, in _new_shared_cuda
    return torch.UntypedStorage._new_shared_cuda(*args, **kwargs)
RuntimeError: CUDA error: invalid resource handle
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

My graphics card is an NVIDIA GeForce RTX 4090 Laptop, which is a GPU chip with some functions trimmed and installed in a laptop. I'm not sure if the problem lies with this GPU chip.

hyacinth531, 2025-06-09:
I have solved this problem, and it still lies in WLS. I was using PyCharm on a Windows operating system to invoke WSL for running WildGS - SLAM. In this configuration, when executing multi - processes and setting the running mode as "torch.multiprocessing.set_start_method('spawn')", the above process conflict problem would occur during multi - process execution. Now I have installed a dual - system on my computer, and directly running it in Ubuntu has solved the problem. Thank you all for your help!

Jianhao-zheng, 2025-06-09:
I am glad to see the problem is solved!

closed 2025-06-09 by Jianhao-zheng

15651888101, 2025-09-05:
@Jianhao-zheng ，May I ask what is your environment? Is it a Ubuntu in VM Workstation or a native linux system？