GradientSpaces/WildGS-SLAM #23: GUI always crushed, and error message is: “OpenGL.GL.shaders.ShaderValidationError: Validation failure (0)”.

dflkasjfh, 2025-06-09:
[image]

We have tried multiple versions of PyOpenGL (from 3.1.0 to 3.1.9), but all of them throw out this problem. Then we try to commend out "program.check_validate()" to see what happened in error, it shows:

[image]

dflkasjfh, 2025-06-11:
Yes, it‘s Ubuntu, and the current version is “Linux omnisky 5.15.0-139-generic #149~20.04.1-Ubuntu SMP Wed Apr 16 08:29:56 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux”.
I have changed to the above versions you provided, but it doesn't work on my computer! 
I have also tried another option to check the online results, but it's not very convenient.
Is there another way to solve this problem better? Thanks!

Jianhao-zheng, 2025-06-30:
Close due to no repsonse for a while