GradientSpaces/WildGS-SLAM #27: Could I replace droid.pth with other models, such as Depth_Anything_V2 or Metric3Dv2 ?

15651888101, 2025-06-23:
Hello Dr. Zheng, I've come to ask questions again.

I'm trying to run the project on my own machine at home(CHN mainland).
Without 'ti zi', I download datatsets with hf-mirror.com, but there is a trouble with the droid.pth on drive.google.com.I haven't found a domestic mirror for it.
By comparing the download links, I have determined that the **droid.pth** you used is from https://github.com/princeton-vl/DROID-SLAM/. In your article, it is mentioned that "In addition, we utilize the metric depth estimated by **Metric3D V2** [13] to stabilize the DBA layer", but in the code, there is thirdparty/**depth_anything_v2**.
I understand that these three methods are all aimed at obtaining metric depth, so there is a possibility of replacement. Is this correct?

Could I replace droid.pth with other models, such as Depth_Anything_V2 or Metric3Dv2 ?
(I have already try it, but they don't include keys "update.weight.2.weight"..., and load_pretrained(in slam.py) failed )

Alternatively, could you suggest a domestic access method for the droid.pth file?  Or a 'ti zi'?

Jianhao-zheng, 2025-06-27:
Hi, thank you for your interest in our work. "ti zi" is VPN in English. **droid.pth** is used to estimated guided flow, not for metric depth estimation. You can not replace it with any depth model. To download that, please find a VPN to access Google. Also, I am a Ph.D. student, not a Dr.

15651888101, 2025-06-28:
Thank you very much, Ph.D. Zheng. I'll think of another way.

closed 2025-06-30 by Jianhao-zheng