GradientSpaces/WildGS-SLAM #29: Request for Uncertainty Model Checkpoint (uncertainty_model.pth)

HanMyDo, 2025-06-23:
Thanks for the great work on WildGS-SLAM! I'm currently experimenting with the uncertainty-based dynamic filtering and noticed that while uncertainty_model.py is included, I couldn't find a pretrained checkpoint (e.g., uncertainty_model.pth) in the repo or scripts_downloading/ folder.
Could you please share a download link or let me know if it's available?

Thanks in advance!

Jianhao-zheng, 2025-06-27:
Hi, thank you for your interest in our work. Our system trains the uncertainty model from scratch for each sequence. It's not a pretrained model. Upon finishing running a sequence, you can find the model weight in ``{output_folder}/uncertainty_mlp_weight.pth"``.

mateoTWD, 2025-07-04:
Is it possible to train the uncertainty model for a scene previously created by a sfm like colmap, pixsfm, fastmap or openmvg and not a SLAM estimation of the parameters of the camera?

Jianhao-zheng, 2025-07-06:
Yes, the same idea can be used in SFM setting. Actually, you can refer to [Nerf on-the-go](https://github.com/cvg/nerf-on-the-go) for  your problem.

closed 2025-07-06 by Jianhao-zheng