GradientSpaces/WildGS-SLAM #31: What's the minimum hardware configuration?

15651888101, 2025-06-28:
Hello, Ph.D. Zheng.
I got droid.pth and solved https://github.com/GradientSpaces/WildGS-SLAM/issues/27.
I encountered the same situation as https://github.com/GradientSpaces/WildGS-SLAM/issues/11. while running Dynamic/Wild_SLAM_iPhone/street.yaml .
Setting metric3d_vit_small doesn't work.
So, What's the minimum hardware configuration?
My environment is:

[image]

Could it be that the system memory and graphics card memory are too small?

Jianhao-zheng, 2025-06-30:
Hi, our experiments are conducted by a 4090 with 16 GB. 8GB is not enough. ORB-SLAM3 is a CPU-only SLAM system.

15651888101, 2025-06-30:
Okay, thanks !