GradientSpaces/WildGS-SLAM #33: Regularization vs Masking of dynamic objects

HaedamOh, 2025-07-23:
Thanks for the great work, 

I am wondering how different the performance between your method 'uncertainty aware regularisation' and simple masking out dynamic objects. 

Is there any particular advantage if you regularise by uncertainty over putting a mask on that area while training ?

Thank you so much

Jianhao-zheng, 2025-07-24:
Hi, thank you for your interest in our work! 

I am a bit confused about your question. Our input is just a sequence of images. There are no GT dynamic masks. We train an uncertainty MLP online to predict the moving objects/humans in 2D frames, namely the uncertainty map. If you have the ground truth dynamic masks, you can simply filter out dynamic pixels in BA and map optimization. But I don't think the GT dynamic mask is easy to achieve.

HaedamOh, 2025-07-26:
Yes, I understood that you try to train uncertainty map as well as RGB. 
But my question was what if you don't train uncertainty map separetly, but just train RGB only with dynamic object masked out while you train. Would not this approach work if our goal is to remove dynamic objects ?

Thank you

Zzh2000, 2025-07-26:
Hi, if I understand correctly, you mean that if we already have a dynamic mask, we can simply apply Gaussian mapping to the non-dynamic pixels? However, obtaining a perfect dynamic mask is not always straightforward. Using semantic segmentation or object detection methods requires class priors, but in many cases, dynamic objects may not even be covered by the predefined classes in models like YOLO. Moreover, the performance of these models is not always reliable, especially in complex/occlusion/partial observation scenarios.

HaedamOh, 2025-07-27:
Thanks for the reply,
Yes, initially I assumed you can get good dynamic masks.
But, now I understand that obtaining dynamic mask itself was a part of challenge. It's clear.

Thank you

closed 2025-07-27 by HaedamOh