GradientSpaces/WildGS-SLAM #36: CUDA Error for Blackwell (RTX5090) GPU

lian-jiarong, 2025-07-31:
Hello, I encountered the error "RuntimeError: CUDA error: no kernel image is available for execution on the device". Are there any debug suggestions for Blackwell GPUs? My CUDA version is 12.8, and the torch version is 2.7.1. Thanks!

QQAndYY-cyber, 2025-08-01:
I encountered the same problem as you; may I ask if you have found a solution?

Jianhao-zheng, 2025-08-01:
Hi, I don't have a 5090 on hand to test. But I think you need to add:

```
'-gencode=arch=compute_120,code=sm_120',
'-gencode=arch=compute_120,code=compute_120',
```
in `setup.py`: https://github.com/GradientSpaces/WildGS-SLAM/blob/498c768911ce6a4a249ff14db603a0b37d1d66a0/setup.py#L26

Please note that the same thing should be added to other `setup.py` of  the thirdparty package.

In addition, since the CUDA version is very new, you probably also need to change something in the lietorch source code.

panyangis, 2025-09-17:
`python -c "import torch; import lietorch; import simple_knn; import diff_gaussian_rasterization; print(torch.cuda.is_available())"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/media/useradmin/B32C-9D8F/SLAM project/WildGS-SLAM/thirdparty/lietorch/lietorch/__init__.py", line 2, in <module>
    from .groups import LieGroupParameter, SO3, RxSO3, SE3, Sim3, cat, stack
  File "/media/useradmin/B32C-9D8F/SLAM project/WildGS-SLAM/thirdparty/lietorch/lietorch/groups.py", line 5, in <module>
    from .group_ops import Exp, Log, Inv, Mul, Adj, AdjT, Jinv, Act3, Act4, ToMatrix, ToVec, FromVec
  File "/media/useradmin/B32C-9D8F/SLAM project/WildGS-SLAM/thirdparty/lietorch/lietorch/group_ops.py", line 1, in <module>
    import lietorch_backends
ImportError: libcudart.so.11.0: cannot open shared object file: No such file or directory
`  
**Hello, I have configured a 5060ti graphics card and used CUDA12.8. Now I have added the following code to WildGS-SLAM/setup.py and WildGS-SLAM/thirdparty**

`'-gencode=arch=compute_120,code=sm_120',
'-gencode=arch=compute_120,code=compute_120',`

**But the error still occurs**

`python -c "import torch; import lietorch; import simple_knn; import diff_gaussian_rasterization; print(torch.cuda.is_available())"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/media/useradmin/B32C-9D8F/SLAM project/WildGS-SLAM/thirdparty/lietorch/lietorch/__init__.py", line 2, in <module>
    from .groups import LieGroupParameter, SO3, RxSO3, SE3, Sim3, cat, stack
  File "/media/useradmin/B32C-9D8F/SLAM project/WildGS-SLAM/thirdparty/lietorch/lietorch/groups.py", line 5, in <module>
    from .group_ops import Exp, Log, Inv, Mul, Adj, AdjT, Jinv, Act3, Act4, ToMatrix, ToVec, FromVec
  File "/media/useradmin/B32C-9D8F/SLAM project/WildGS-SLAM/thirdparty/lietorch/lietorch/group_ops.py", line 1, in <module>
    import lietorch_backends
ImportError: libcudart.so.11.0: cannot open shared object file: No such file or directory
`
**I sincerely ask you what source code needs to be modified**